takkaO/OpenFontRender #1: 複数のフォントを表示できるようにしたい

alto0126, 2022-02-23:
下記のように複数のOpenFontRenderクラスから別のインスタンスを生成し、それぞれフォントファイルをloadFont()で指定して、それぞれを表示しようとしましたが、最後にloadFont()した物しか表示されません。
テストした、ArduinoIDEのコードを載せます。

#include "M5Core2.h"
#include "OpenFontRender.h" 

void setup()
{
  M5.begin();
    OpenFontRender ofr1;
    OpenFontRender ofr2;
    M5.Lcd.fillScreen(BLACK);
    ofr1.loadFont("/font/OpenSans-Light-webfont.ttf");
    ofr2.loadFont("/font/OpenSans-Bold.ttf");
    ofr1.setDrawer(M5.Lcd);
    ofr1.setFontSize(30);
    ofr1.setCursor(0, 0);
    ofr1.printf("ABCDEFG");
    //ofr2.loadFont("/font/OpenSans-Bold.ttf");
    ofr2.setDrawer(M5.Lcd);
    ofr2.setFontSize(30);
    ofr2.setCursor(0, 50);
    ofr2.printf("12345678");
}

void loop() {
  // put your main code here, to run repeatedly:
}

[fonttest.zip](https://github.com/takkaO/OpenFontRender/files/8122284/fonttest.zip)

takkaO, 2022-02-27:
検証途中ですが、進展があったので報告します。

### フォントファイル容量が大きいため、ロードに失敗した可能性
サブセットフォントメーカーでASCIIのみに軽量化したフォントで検証した結果、症状改善せず。
よってフォントファイル容量が原因ではない

### フォントのコード埋め込み時でも同様の症状が出るか
```ttf2bin.py``` によってコード埋め込みにした場合、両フォント（軽量化済み）は正常に表示。
よって、ファイルからの読み込み時にのみ症状が出る

以上の検証結果から、原因はファイル読み込みの部分だと考えられます。
おそらくファイル読み込み部分の処理がインスタンスごとに分けられていないため、
後からロードしたフォントで上書きされてしまっているのだと思います。

この問題に対応する予定ですが、おそらく時間がかかります。
しばらくの間、複数フォント利用時はコード埋め込みによる利用をしていただければと思います。

alto0126, 2022-02-27:
私の方でもその後色々調べていますが原因は見つけられていません。私もファイルアクセス辺りが怪しいと考えています。
例えば、下記の様にコードを書いてみると

    ofr1.loadFont("/font/ShipporiMincho-Bold.ttf");
    ofr1.setDrawer(M5.Lcd);
    ofr1.setFontSize(30);
    ofr1.setCursor(0, 0);
    ofr1.printf("1234");
    ofr2.loadFont("/font/OpenSans-Bold.ttf");
    ofr2.setDrawer(M5.Lcd);
    ofr1.setFontSize(30);
    ofr1.setCursor(0, 50);
    ofr1.printf("123456789");
    ofr2.setFontSize(30);
    ofr2.setCursor(0, 100);
    ofr2.printf("12345678");
    
    最初にofr1で、「1234]と出力したときにこの内容はキャッシュに残っているようで、その後、ofr2.loadFont()を実行した後に、ofr1で「123456789」と出力すると、[1234」まで正常に出力されて、それ以降はエラーで表示されなくなります。キャッシュにフォントが無いときファイルに読みに行く辺りが原因かと疑っていますが、そこまでソースが追えていません。
    
コードへの埋め込みはまだ試していません。

takkaO, 2022-03-06:
対策を行ったバージョンをアップしました。
SDカードからの読み込みでも複数フォントを正常に表示できるようになったはずです。
お試しください。

alto0126, 2022-03-07:
バージョンアップ版で確認しました。３つのフォントを表示して問題なく表示されています。
綺麗なフォント表示ができるようになりました。ありがとうございました。

http://remote.beruf.xyz/m5stack/P1030749s.jpg

closed 2022-03-07 by alto0126