takkaO/OpenFontRender #3: Seems unable to render fonts with too large font size

HalfSweet, 2022-05-13:
Hello, I came across your project and I ported it to my own device, excitingly it works for the most part, but I found a frustrating problem. When I set the font too big (about bigger than 170) it doesn't draw the font I want, have you encountered a similar problem?
This is my code,I did not find any errors.
```c++
if (render->loadFont(_TimeFontPath))
	{
		Serial.println("Render initialize error");
		return;
	}
render->setDrawPixel(EPD->DrawPixel);
render->setFontSize(180);
render->setCursor(50, 70);
render->printf("A");
render->drawChar('1', 100, 100);
```

takkaO, 2022-05-14:
Thank you for using this library! 😄

The same problem occurred in my environment too.
I will try to fix it. Please wait for a while.

takkaO, 2022-11-24:
@HalfSweet 

The issue has been fixed and merged into the master branch.
Thank you for your contribution.

P.S.
Sorry for the delay 🙇‍♂️

MikeyMoMo, 2023-01-16:
I have displayed up to 180 points and it works great.  I call it fixed!

closed 2023-02-17 by HalfSweet