takkaO/OpenFontRender #4: Possible to add a font range? Generated font with ttf2bin > 24 MB

martinberlin, 2022-10-10:
Hi @takkaO 
Very interesting project I'm thinking to use it in combination with Lovyan GFX to use chinese fonts in a demo Firmware. 

One question: Is possible to add size / range to ttf2bin.py?

When I try to generate a binary for [this chinese font](https://chinesefonts.org/fonts/yrdzst-semibold):

3.9M Jun 27  2021 yrd_b.ttf
24M Oct 10 13:12 binaryttf.h

I'm getting a 24 megabytes header file. There is no way that I can use that in a microcontroller.
What's the way to generate smaller fonts?

martinberlin, 2022-10-10:
Thanks!
The problem is that I use Linux and that program is only for Windows or Macintosh.

But I found this one called FontForge:
https://www.linuxadictos.com/en/fontforge-un-editor-de-fuentes-tipograficas-multiplataforma.html

Will try to check if it's possible to extract a subset of Fonts. One question: Is not possible to make it smaller reducing the size of the Glyph?

On other GFX tools like Adafruit GFX you can pass the size to the font convert command.

martinberlin, 2022-10-10:
Update this Font Forge seems to work correctly in Linux. Now I just need to understand how I will write one of that characters in unicode and it will automatically find it with your library.
Made a fork of the library and added support for Espressif IDF framework:

https://github.com/martinberlin/OpenFontRender-idf

Needed to remove some Serial.prints (Those are supported only in Arduino)
Excluded Arduino.h when a __LINUX__ define is set. The rest where minimal touches and added a CMakeLists file that is needed by IDF components.

takkaO, 2022-10-11:
I'm glad it worked out. 🎉

`OpenFontRender` is currently undergoing an update process in the develop branch to make it more functional and less Arduino-dependent. (The reduction of Arduino dependencies includes moving away from Serial.)

The main code update has already been completed, and I am adding sample code and fixing documentation. 
New `OpenFontRender` will be released soon.
If you can, please try the develop branch at ESP-IDF and give us your feedback, it would be greatly appreciated.

Finally, in your question `Is not possible to make it smaller reducing the size of the Glyph?` do you mean the drawing font size?
If so, you can change the font size with `setFontSize` method.

Thank you.

martinberlin, 2022-10-11:
Great!
Feel free to grab the CMakeList files from my fork. Arduino will ignore it, but CMake based compilers like Espressif IDF will use it to build the component.
I will try the develop branch when it’s ready.

one more question that I still don’t know how it goes:
How does open font render matches a Chinese glyph ex temperature 

So temperature today here is: 

21.9 温度 C.  76.3 % 湿度  76.3 % 湿度 humidity 

How it finds the glyph in that big array?
Or should I organize them like :
1 = 温
2 = second glyph, etc.

I hope my question is understandable. Basically what I mean is:
How I need to organize the subset of the font in order that is possible to use and that the Asian glyph is matched?

martinberlin, 2022-10-11:
Quick question is there a way to invert the background/foreground color?
[image]
I still didn’t get how to get the same gray color I’m using in LovyanGFX GFX with
Display.color888(255,255,255) that should be white in RGB but is an uint32 so maybe is getting lost when converter to uint16 in render.setFontColor

martinberlin, 2022-10-12:
Hi @takkaO 
I tried to build develop branch in IDF v5. and this is the first error I found:

```
1.
../components/OpenFontRender/upstream/src/OpenFontRender.cpp:496:44: error: 'unicode' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  496 |         if (detect_control_char && unicode == '\n') {
      |                                    ~~~~~~~~^~~~~~~
//1 simply solved adding an initializer value: uint16_t unicode = 0; // Line 277
```

martinberlin, 2022-10-13:
Thanks for the help on this one. Closing since the IDF fix is being handled in the commit comments

closed 2022-10-13 by martinberlin