takkaO/OpenFontRender #5: How easy is to add a getTextWidth(char* text_buffer) function?

martinberlin, 2022-10-12:
To design stuff sometimes is great to know how much is the text width (Using font size) that you will draw. Lovyan has a function like this:

```C
uint16_t getTextWidth(char* text_buffer);
```

@takkaO if you give me some pointers on how to do it I can add it myself.
On another question: Is possible to speed up just a little bit the drawing?
Tried with ESP32 at 80Mhz and also at 160Mhz, the difference is not noticeable, but still I like a lot to see how it draws the font from the binary TTF. Amazing work!

Bodmer, 2022-11-11:
I have issued a pull request that speeds up rendering. This should work with any Adafruit_GFX API compatible library (drawFastHLine fn).

Bodmer, 2022-11-11:
ESP32 renders example "load from binary" now reports 54ms (8 bit parallel interface). If the cache is valid for a repeat render then this reduces to 27ms, which is circa 1ms per character.

martinberlin, 2022-11-11:
That's awesome. I'm using this primarily in some IT8951 (parallel epaper displays) but I'm sending data from ESP32S3 as fast I can per SPI. Will try and see if I can get more speed with your fix.
    But I'm using @lovyan03 Lovyan GFX as a controller library and they don't use Adafruit but their own GFX.

takkaO, 2022-11-23:
@martinberlin 

`getTextWidth` and `getTextHeight` methods have been added.
However, if you need both width and height, it is recommended that you use the `calculateBoundingBox` or `calculateBoundingBoxFmt` method and get the bounding box instead.
You can calculate width and height from bounding box and it is less computationally intensive that way.

martinberlin, 2022-11-24:
Awesome work will try it before closing the ticket to give you feedback.
Thanks a lot!

martinberlin, 2023-01-13:
Thanks TakkaO working as expected. Still didn't try the calculateBoundingBox but I will soon!

closed 2023-01-13 by martinberlin