takkaO/OpenFontRender #6: Compile fails for ESP32 in Arduino environment

Bodmer, 2022-11-09:
Compile fails for the ESP32, warnings and errors are reported below. This is with the ESP32 board package 2.0.5 which seems to be very picky with type conversions:
```
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp: In member function 'FT_Error OpenFontRender::loadFont(const unsigned char*, size_t)':
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:201:59: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  error = FTC_Manager_LookupFace(_ftc_manager, (FTC_FaceID)_face_id, &face);
                                                           ^~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp: In member function 'FT_Error OpenFontRender::loadFont(const char*)':
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:244:59: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  error = FTC_Manager_LookupFace(_ftc_manager, (FTC_FaceID)_face_id, &face);
                                                           ^~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp: In member function 'FT_Error OpenFontRender::drawChar(uint16_t, uint32_t, uint32_t, uint16_t, uint16_t)':
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:279:59: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
  error = FTC_Manager_LookupFace(_ftc_manager, (FTC_FaceID)_face_id, &face);
                                                           ^~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:286:57: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
                                             (FTC_FaceID)_face_id,
                                                         ^~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:331:36: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   image_type.face_id = (FTC_FaceID)_face_id;
                                    ^~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp: In member function 'uint16_t OpenFontRender::drawString(const char*, uint32_t, uint32_t, uint16_t, uint16_t, FT_Error*)':
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:382:61: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   *error = FTC_Manager_LookupFace(_ftc_manager, (FTC_FaceID)_face_id, &face);
                                                             ^~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp: In function 'void RenderTask(void*)':
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:638:79: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
                                                   (FTC_FaceID)g_TaskParameter.face_id,
                                                                               ^~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:646:52: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
    ft_scaler.face_id = (FTC_FaceID)g_TaskParameter.face_id;
                                                    ^~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:676:28: error: narrowing conversion of 'face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::bitmap.FT_Bitmap_::width' from 'int' to 'FT_Byte' {aka 'unsigned char'} inside { } [-Werror=narrowing]
        face->glyph->bitmap.width,      // width
        ~~~~~~~~~~~~~~~~~~~~^~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:677:28: error: narrowing conversion of 'face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::bitmap.FT_Bitmap_::rows' from 'int' to 'FT_Byte' {aka 'unsigned char'} inside { } [-Werror=narrowing]
        face->glyph->bitmap.rows,       // height
        ~~~~~~~~~~~~~~~~~~~~^~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:678:21: error: narrowing conversion of 'face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::bitmap_left' from 'FT_Int' {aka 'int'} to 'FT_Char' {aka 'signed char'} inside { } [-Werror=narrowing]
        face->glyph->bitmap_left,       // left
        ~~~~~~~~~~~~~^~~~~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:679:21: error: narrowing conversion of 'face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::bitmap_top' from 'FT_Int' {aka 'int'} to 'FT_Char' {aka 'signed char'} inside { } [-Werror=narrowing]
        face->glyph->bitmap_top,        // top
        ~~~~~~~~~~~~~^~~~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:681:28: error: narrowing conversion of 'face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::bitmap.FT_Bitmap_::num_grays' from 'short int' to 'FT_Byte' {aka 'unsigned char'} inside { } [-Werror=narrowing]
        face->glyph->bitmap.num_grays,  // max_gray
        ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:682:28: error: narrowing conversion of 'face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::bitmap.FT_Bitmap_::pitch' from 'int' to 'FT_Short' {aka 'short int'} inside { } [-Werror=narrowing]
        face->glyph->bitmap.pitch,      // pitch
        ~~~~~~~~~~~~~~~~~~~~^~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:683:31: error: narrowing conversion of '(face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::advance.FT_Vector_::x >> 6)' from 'FT_Pos' {aka 'long int'} to 'FT_Char' {aka 'signed char'} inside { } [-Werror=narrowing]
        face->glyph->advance.x >> 6,    // xadvance
        ~~~~~~~~~~~~~~~~~~~~~~~^~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:684:31: error: narrowing conversion of '(face->FT_FaceRec_::glyph->FT_GlyphSlotRec_::advance.FT_Vector_::y >> 6)' from 'FT_Pos' {aka 'long int'} to 'FT_Char' {aka 'signed char'} inside { } [-Werror=narrowing]
        face->glyph->advance.y >> 6,    // yadvance
        ~~~~~~~~~~~~~~~~~~~~~~~^~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:695:53: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
    image_type.face_id = (FTC_FaceID)g_TaskParameter.face_id;
                                                     ^~~~~~~
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp: In function 'FT_Error ftc_face_requester(FTC_FaceID, FT_Library, FT_Pointer, FT_FaceRec_**)':
D:\XXXX\Arduino\Sketches\libraries\OpenFontRender\src\OpenFontRender.cpp:566:9: error: 'error' may be used uninitialized in this function [-Werror=maybe-uninitialized]
  return error;
         ^~~~~
cc1plus.exe: some warnings being treated as errors
```

Bodmer, 2022-11-09:
When a RP2040 based board is selected the code compiles, uploads and runs correctly. So this compile problem seems to be associated with the ESP32 board package.

martinberlin, 2022-11-09:
Hello Boner, try to switch the branch to develop.
I think they are updates there that are not yet merged in master branch

Bodmer, 2022-11-09:
I have found that in the develop branch I have to comment out the following lines in the sketch:

//  render.showFreeTypeVersion(); // print FreeType version
//  render.showCredit();      // print FTL credit

Each line causes the RP2040 processor to crash if it is uncommented. I have not investigated further.

The rendering of the characters in the delvelop branch to the TFT is also significantly slower, I have not investigated why this may be the case.

Bodmer, 2022-11-09:
I looked at this further and it is g_Print() in OpenFontRender.h/cpp that is causing the RP2040 processor crash with the develop branch.

I am guessing this has to be user defined or is expected to be part of the debug environment used during development. This needs to be resolved.

Bodmer, 2022-11-09:
I added the following to the sketch and ir now runs using the develop branch without crashing on the RP040.
```
  Serial.begin(115200);
  Serial.flush();
  render.setSerial(Serial); // Important to avoid a crash with RP2040
```

Bodmer, 2022-11-10:
Thank you, it's a great easy to use library. It is working very well now on my RP2040 with those tweaks.
I have created new binaries from ttf files without a problem.

Bodmer, 2022-11-11:
I have issued a pull request based on the develop branch that fixes this.

takkaO, 2022-11-24:
Pull requests have been merged and reflected in the master branch.
Thanks for your contribution 👍

Bodmer, 2022-11-24:
Great, happy to help.

closed 2022-11-24 by Bodmer