takkaO/OpenFontRender #12: unloadFont crashes ESP32 S3 when font file is in FLASH

Bodmer, 2022-11-28:
The ESP32 and RP2040 processors run fine, but the new ESP32 S3 processor crashes after using a font file in FLASH and then calling unloadFont().

[This line](https://github.com/takkaO/OpenFontRender/blob/master/src/OpenFontRender.cpp#L249)  is causing the crash. Since the sketch has provided the pointer in this case it seems inappropriate for the library to try to delete the memory allocation.

takkaO, 2022-11-28:
Hi @Bodmer 

The line that you teach me has been removed in the latest develop branch. 
However, we have not checked if it really does not crash.

If you don't mind, could you please provide feedback on whether this issue has been resolved in the develop branch ?

Thank you.

Bodmer, 2022-11-28:
Deleting the line fixes it.

trailsurfer604, 2022-12-23:
This fix worked for me too: I deleted line 249 from the library and my program stopped crashing on font unload.

Bodmer, 2022-12-23:
Other changes suggested as in #16

MikeyMoMo, 2023-01-03:
I just commented out line 249 and recompiled everything and successfully got rid of the abort.  Here is the info from before the change.

assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x400e889b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40089020:0x3ffb1e20 0x40089299:0x3ffb1e40 0x400e889b:0x3ffb1e60 0x40081a07:0x3ffb1e90 0x40085b79:0x3ffb1eb0 0x4000bec7:0x3ffb1ed0 0x400feb89:0x3ffb1ef0 0x400fe8f1:0x3ffb1f10 0x400d736e:0x3ffb1f30 0x400d170f:0x3ffb1f50 0x400e649d:0x3ffb1fb0 0x4008a29e:0x3ffb1fd0
---------------------------------------------------------------------------------------
0x40089020: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40089299: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400e889b: __assert at ../../../.././newlib/libc/stdlib/assert.c line 74
0x40081a07: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x40085b79: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x400feb89: read_encoded_value_with_base(unsigned char, _Unwind_Ptr, unsigned char const*, _Unwind_Ptr*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/../libgcc/unwind-pe.h line 201
0x400fe8f1: __gnu_cxx::__concurrence_unlock_error::what() const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/ext/concurrence.h line 80
0x400d736e: OpenFontRender::unloadFont() at D:\Arduino\libraries\OpenFontRender-master\src\OpenFontRender.cpp line 250
0x400d170f: setup() at C:\Users\admin\AppData\Local\Temp\arduino_modified_sketch_727637/Sprite_Noto_Font_Demo.ino line 75
0x400e649d: app_main() at C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 29
0x4008a29e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143