takkaO/OpenFontRender #13: Unable to use OpenFrontRender to use with TFT-eSPI

buckfast-beekeeper, 2022-11-28:
I use the example load_from_binary.ino with a little adjustments. 

```
#include "TFT_eSPI.h" // tryed <TFT_eSPI.h> too
#include "binaryttf.h" 
#include "OpenFontRender.h" // tryed <OpenFontRender.h> too

#define BACKLIGHT_PIN 13

TFT_eSPI tft;
OpenFontRender render;

void setup() {
  Serial.begin(19200);
  analogWrite(BACKLIGHT_PIN, 200);
  analogWriteFreq(250);

  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLUE);
  tft.setCursor(10,100);
  tft.println("Test");
  delay(2000);  // until here it works perfect

  if (render.loadFont(binaryttf, sizeof(binaryttf))) {
    Serial.println("Render initialize error");
    return;
  }
  render.setDrawer(tft); // Set drawer object
  tft.fillScreen(TFT_WHITE);
  render.setFontColor(TFT_WHITE, TFT_BLUE);
  render.printf("Hello World\n");
  render.seekCursor(0, 10);
  delay(1000);

  render.setFontSize(30);
  render.setFontColor(TFT_GREEN, TFT_RED);
  render.printf("完全なUnicodeサポート\n");
  render.seekCursor(0, 10);
  delay(1000);

  render.setFontSize(40);
  render.setFontColor(TFT_ORANGE, TFT_BLACK);
  render.printf("こんにちは世界\n");
}

void loop() {
}
```

binaryttf.h is in the same map as the program. All works fine until the font is loaded. 

I use these this 

```
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// User defined information reported by "Read_User_Setup" test & diagnostics example
#define USER_SETUP_ID 500

// Define to disable all #warnings in library (can be put in User_Setup_Select.h)
//#define DISABLE_ALL_LIBRARY_WARNINGS

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

#define ILI9486_DRIVER

#define TFT_INVERSION_OFF
//#define TFT_INVERSION_ON

// ##################################################################################
//
// Section 2. Define the pins that are used to interface with the display here
//
// ##################################################################################

// ###### EDIT THE PIN NUMBERS IN THE LINES FOLLOWING TO SUIT YOUR ESP8266 SETUP ######

// For NodeMCU - use pin numbers in the form PIN_Dx where Dx is the NodeMCU pin designation
#define TFT_CS   PIN_D1  // Chip select control pin D8
#define TFT_DC   PIN_D2  // Data Command control pin
#define TFT_RST  PIN_D3  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TFT_RST  -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

//#define TFT_BL PIN_D1  // LED back-light (only for ST7789 with backlight control pin)

//#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen
#define TOUCH_CS -1

// ##################################################################################
//
// Section 3. Define the fonts that are to be used here
//
// ##################################################################################

// Comment out the #defines below with // to stop that font being loaded
// The ESP8366 and ESP32 have plenty of memory so commenting out fonts is not
// normally necessary. If all fonts are loaded the extra FLASH space required is
// about 17Kbytes. To save FLASH space only enable the fonts you need!

//#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
//#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
//#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
//#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
//#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
//#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

// ##################################################################################
//
// Section 4. Other options
//
// ##################################################################################
//#define SPI_FREQUENCY  27000000
 #define SPI_FREQUENCY  40000000
// #define SPI_FREQUENCY  55000000 // STM32 SPI1 only (SPI2 maximum is 27MHz)
// #define SPI_FREQUENCY  80000000

// Optional reduced SPI frequency for reading TFT
//#define SPI_READ_FREQUENCY  20000000

// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
//#define SPI_TOUCH_FREQUENCY  2500000
```

takkaO, 2022-11-28:
Hi @buckfast-beekeeper .
Thank you for using this library 😄

I need more information to solve the problem.

- What microcontroller are you using?
- What development environment and version are you using?
- Is your program black out? crash and reboot?
- What is the output to the serial monitor?
- Have you checked that TFT_eSPI works by itself, i.e., that the `drawPixel` and `drawFastHLine` functions work correctly?

Thank you.

buckfast-beekeeper, 2022-11-28:
Many thanks for your time and librarie. 

I use an ESP8266 Wemos 1 Pro.
Arduino IDE 1.8.19 but I can try IDI 2.0.
`tft.fillScreen(TFT_WHITE);`
Is the last thing that is executed. No reboot, 
On the serial monitor nothings happens. 
TFT_eSPI works fine. If I take the examples/480 x 320/Demo_3D_Cube or TFT_rainbow480, they works like excepted. 

I have try the littleFS examples with .vlw fonts, they worked too.

buckfast-beekeeper, 2022-11-28:
Update: The program works great with ESP32 DEV board and ILI9486 AND with ILI9341

It seems the problem is with the ESP8266.

martinberlin, 2022-11-28:
I would check how much free heap there is. It's ages I don't program one, but I think have some in the deep of my drawer, what I don't have is an ILI9486...But just tell me if I can help debugging something

takkaO, 2022-11-29:
I don't know if this will solve the problem, but try to add the following code at the beginning of the setup function.

```c++
void setup() {
    ...
    render.setSerial(Serial);    // Call after Serial.begin()
    render.setDebugLevel(OFR_DEBUG);    // optional
    ...
}
```

Then check the serial output.

buckfast-beekeeper, 2022-11-30:
This is the font setup:
```
void setupFont() {
  Serial.println("Setup Font");

  ofr.setSerial(Serial);    // Need to print render library message
  ofr.setDebugLevel(OFR_DEBUG);    // optional
  //ofr.showFreeTypeVersion(); // print FreeType version
  //ofr.showCredit();
  ofr.setDrawer(tft);// Link drawing object to tft instance (so font will be rendered on TFT)
  ofr.setFontColor(textColor, backColor);

  //ofr.loadFont(TTF_FONT, sizeof(TTF_FONT));
  if (ofr.loadFont(TTF_FONT, sizeof(TTF_FONT))) {
    Serial.println("Initialise error");
    return;
  }
  ofr.setCursor(10, 200);
  ofr.setFontSize(100);
  //ofr.rprintf("zon 27 nov 2022");
  //ofr.rprintf("zon 27 nov 2022");
  tft.println("zon 27 nov 2022");
  ofr.setFontColor(textColor, backColor);
  ofr.cprintf("Hello World\n");
  ofr.seekCursor(0, 10);

  Serial.println("Setup Font OK");
}
```

This is the Serial output/
```
10:36:29.374 -> Setup OLED
10:36:30.901 -> Setup Font
10:36:30.901 -> Setup Font OK
```
After compiling this is the message in the arduino
```
Bibliotheek TFT_eSPI op versie 2.4.79 in map: E:\Arduino\libraries\TFT_eSPI  wordt gebruikt
Bibliotheek SPI op versie 1.0 in map: C:\Users\gebruiker\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SPI  wordt gebruikt
Bibliotheek OpenFontRender-master op versie 1.0.0 in map: E:\Arduino\libraries\OpenFontRender-master  wordt gebruikt
De schets gebruikt 414289 bytes (39%)  programma-opslagruimte. Maximum is 1044464 bytes.
Globale variabelen gebruiken 40208 bytes (49%) van het dynamisch geheugen. Resteren 41712 bytes voor lokale variabelen. Maximum is 81920 bytes.
```
```
Serial port COM24
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 40:91:51:51:01:3a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 418448 bytes to 300178...
```

takkaO, 2022-11-30:
Still having problems with display?

Sorry, I missed the debugLevel argument 🙇‍♂️. 
Please re-try below.

```c++
ofr.setDebugLevel(OFR_DEBUG | OFR_INFO | OFR_ERROR);
```

takkaO, 2022-11-30:
Great ! 🎉
We've succeeded in finding a source of the problem!

[According to the FreeType site](https://freetype.org/freetype2/docs/reference/ft2-error_code_values.html), error code 0x82 mean a stack overflow error.

buckfast-beekeeper, 2022-11-30:
With a [Lolin Wemos D1 pro](https://www.wemos.cc/en/latest/d1/d1_mini_pro.html) with 16MB flash, same error.

takkaO, 2022-11-30:
I think 16MB mean Flash memory size.
But what we need is RAM size.

According to my lightly researched information, ESP8266 have about 50 kB RAM.
The ESP32 has about 520 kB, the WioTerminal about 192 kB, and the RP2040 about 264 kB of RAM.

Adding an external SRAM to the ESP8266 might work.
(But it would be much more economical and easier to just use the ESP32.)

buckfast-beekeeper, 2022-12-01:
> Adding an external SRAM to the ESP8266 might work.
> (But it would be much more economical and easier to just use the ESP32.)

I have multiple ESP32 in stock. That isn't the problem. I was thinking I use only 7 pins, why taking the ESP32 with more pins? 

Biggest consequence is remake the PCB and reprint the enclosure.

martinberlin, 2022-12-01:
That is the issue then. As @takkaO mentioned the 8266 has around 50 Kb.
And seems after loading the font only 18 Kb are free. Impossible to keep on doing anything with such a low Heap, since the Espressif chips have to do also WiFi stuff on their own, and usually crash quickly having not enough heap RAM.
The best would be to use ESP32. Or if you want to use a MCU that is still recommended for new designs (All esp32 are NRND in their datasheets at this point) then go for a mini ESP32S3.

buckfast-beekeeper, 2022-12-01:
The heap size is without WiFi started. Only TFT_eSPI and OpenFontRender is loaded. 

I have around 20 ESP32 in stock. No direct problems with that and at the moment they are available without problems. I don't make great series. If they are sold out, I can take another, maybe more recently µC.

Bodmer, 2022-12-12:
Just noticed this post. I will update the ReadMe in TFT_eSPI to reflect this.

takkaO, 2022-12-14:
@buckfast-beekeeper 

We found that the cause of this issue is due to heap memory.
A change in the FreeType version or options, or a code fix in OpenFontRender could improve the problem with ESP8266.
Unfortunately I cannot make any further code corrections because I do not have an ESP8266.
Therefore, I am going to close this issue.

If someone can provide a solution or fix code to this problem, I would welcome it.

closed 2022-12-23 by takkaO