takkaO/OpenFontRender #14: Meaning of error code

jie326513988, 2022-12-02:
Hi
When I use loadFont, it returns FTC_ Manager_ LookupFace error: 0x02, but I can't find the document to explain the meaning of the error code. Can you help me.

takkaO, 2022-12-02:
Hi @jie326513988 
Thank you for using this library 😊

Error codes are from the FreeType library.
[FreeType Errors](https://freetype.org/freetype2/docs/reference/ft2-error_code_values.html)

0x02 mean `unknown file format`, your font file is not True Type Font.

jie326513988, 2022-12-02:
> 嗨@jie326513988感谢您使用此库😊
> 
> 错误代码来自 FreeType 库。[自由类型错误](https://freetype.org/freetype2/docs/reference/ft2-error_code_values.html)
> 
> 0x02的意思是，您的字体文件不是真实类型字体。`unknown file format`

Thank you very much. Your library is great

jie326513988, 2022-12-02:
I found that the problem was caused by the failure to pass in the file name.
Serial port output:
14:03:27.958 -> loadFont:0
14:03:27.958 -> [OFR ERROR] Font load Failed: 0x02
14:03:27.958 -> [OFR ERROR] FTC_ Manager_ LookupFace error: 0x02
14:03:27.958 -> Render initialize error
Code snippet:
if (!sdBeginCheck()) esp_ sleep(0);
char* fontName = "/Zfull_GB.ttf";
if (render.loadFont(fontName))
{
Serial. println("Render initialize error");
esp_ sleep(0);
return;
}
[https://github.com/jie326513988/EPD_PixelJitter_BMP_JPG/blob/main/20221202140637.png](url)

closed 2022-12-02 by jie326513988