takkaO/OpenFontRender #15: Text rendered below the cursor?

trailsurfer604, 2022-12-22:
Perhaps this is a feature rather than a bug, but I am running in to an issue that text is being rendered below the cursor (i.e. under the Y coordinates), rather than above.  

I am using an ILI9341 240x320 display

Here are my text coordinates, and calls to the printf() function:

`//Measurements display - numbered    1 2 / 3 4 
const unsigned char * MEAS_TEXT_FONT = NotoSans_Bold;
const uint8_t MEAS_TEXT_FONT_SIZE = 24;
const unsigned char * MEAS_VAL_FONT = NotoSans_Regular;
const uint8_t MEAS_VAL_FONT_SIZE = 24;

const uint8_t MEAS1_TEXT_X     = 20;
const uint8_t MEAS1_TEXT_Y     = 75;
const char * MEAS1_TEXT = "Boost";

const uint8_t MEAS1_VAL_X    = 46;
const uint8_t MEAS1_VAL_Y    = 123;
const char * MEAS1_VAL_UNIT = " PSI";
double boost = 0;

const uint8_t MEAS2_TEXT_X     = 193;
const uint8_t MEAS2_TEXT_Y     = 75;
const char * MEAS2_TEXT = "EGT";

const uint8_t MEAS2_VAL_X    = 219;
const uint8_t MEAS2_VAL_Y    = 123;
const char * MEAS2_VAL_UNIT = " F";
double egt_temp_f = 0;

const uint8_t MEAS3_TEXT_X     = 20;
const uint8_t MEAS3_TEXT_Y     = 171;
const char * MEAS3_TEXT = "Coolant";

const uint8_t MEAS3_VAL_X    = 46;
const uint8_t MEAS3_VAL_Y    = 219;
const char * MEAS3_VAL_UNIT = " F";
double coolant_temp_f = 0;

const uint8_t MEAS4_TEXT_X     = 193;
const uint8_t MEAS4_TEXT_Y     = 171;
const char * MEAS4_TEXT = "Trans";

const uint8_t MEAS4_VAL_X    = 219;
const uint8_t MEAS4_VAL_Y    = 219;
const char * MEAS4_VAL_UNIT = " F";
double trans_temp_f = 0;
`

`void drawStaticSensorScreen () { 
  
  // draw banner double line
  display.drawFastHLine(1, LINE1_Y, 320, COLOR_WHITE);  // draw banner line
  display.drawFastHLine(1, LINE1_Y+2, 320, COLOR_WHITE);  // draw banner line
  
  // set up four measurement labels

  // Load the font and check it can be read OK
  if (render.loadFont(NotoSans_Bold, sizeof(NotoSans_Bold))) {
    Serial.println("Error Loading MEAS_TEXT_FONT ");
    return;
  }
  render.setFontColor(COLOR_WHITE);
  render.setFontSize(MEAS_TEXT_FONT_SIZE);
  
  render.setCursor(MEAS1_TEXT_X, MEAS1_TEXT_Y);
  render.printf(MEAS1_TEXT);
  
  render.setCursor(MEAS2_TEXT_X, MEAS2_TEXT_Y);
  render.printf(MEAS2_TEXT);
  
  render.setCursor(MEAS3_TEXT_X, MEAS3_TEXT_Y);
  render.printf(MEAS3_TEXT);
  
  render.setCursor(MEAS4_TEXT_X, MEAS4_TEXT_Y);
  render.printf(MEAS4_TEXT);
  
  render.unloadFont();

  // for test purposes, fill in some sample measurements
  // banner measurements
  if (render.loadFont(NotoSans_Regular, sizeof(NotoSans_Regular))) {
    Serial.println("Error Loading MEAS_VAL_FONT ");
    return;
  }

  render.setFontSize(MEAS_VAL_FONT_SIZE);

  render.setCursor(MEAS1_VAL_X, MEAS1_VAL_Y);
  boost = 15;
  render.printf("%.0lf", banner_val1);
  render.printf(MEAS1_VAL_UNIT);

  render.setCursor(MEAS2_VAL_X, MEAS2_VAL_Y);
  egt_temp_f = 700;
  render.printf("%.0lf", egt_temp_f);
  render.printf(MEAS2_VAL_UNIT);

  render.setCursor(MEAS3_VAL_X, MEAS3_VAL_Y);
  coolant_temp_f = 180;
  render.printf("%.0lf", coolant_temp_f);
  render.printf(MEAS3_VAL_UNIT);

  render.setCursor(MEAS4_VAL_X, MEAS4_VAL_Y);
  trans_temp_f = 170;
  render.printf("%.0lf", trans_temp_f);
  render.printf(MEAS4_VAL_UNIT);
  
  render.unloadFont();
}`

Here is the result I am getting.  
[image]

And here is is what I should be getting with the same coordinates (displayed using u8g2 library)
[image]

takkaO, 2022-12-23:
Hi @trailsurfer604
Thank you for using this library 😊

In this program, it is drawn from below the cursor (i.e., below the y-coordinate) because it is easier to code.
However, it would be possible to add code to draw above the cursor (i.e., above the y-coordinate).

I am not sure which is more common.
I will try to write some code to verify, please give me time.

Bodmer, 2022-12-23:
A top left datum for text rendering is more common for Arduino TFT libraries. This is convenient for screens since any size text can be rendered at say 0,0 on the screen and other functions such as drawRectangle also use top left for the datum.
This is then compatible with the way other Arduino graphics handle the cursor datum. Note that Adafruit_GFX uses the top left for the basic GLCD font and set the trend for this, but then broke with consistency and used the font baseline for GFX free fonts so it does get confusing.

Ideally the ability to move the datum in the sketch is useful however the font metrics available allow this to be done ing the sketch.

takkaO, 2023-05-01:
More detailed alignment settings are available in `v1.1`. 
Try using the `setAlignment()` function. 
See [GitHub Pages](https://takkao.github.io/OpenFontRender/group__layout__api.html#ga7d72198aa32b966aa60c7cdf54c83d1a) for more information.