takkaO/OpenFontRender #17: If no cache is set, it defaults to CACHE_SIZE_NO_LIMIT which will consume all available (heap) memory

Ramon-Leeuwe, 2023-01-01:
If no cache is set, it defaults to CACHE_SIZE_NO_LIMIT which will consume all available (heap) memory.
It might be better to set the default to a lower limit and add a note in the doc's to explain how to increase the cache if needed.

MikeyMoMo, 2023-01-13:
I am getting a crash related to cache size.  I think I need this fix, too.  Or instructions on how to limit cache usage so I can continue with my project until a permanent fix is issued.  Thanks for a great library.  I hope I can use it fully, soon.  It works once but the second time I try to use it, the program crashes.  Works fine with bitmapped fonts.  Hoping to switch over to TTF because I use multiple sizes and this would reduce memory usage.  Thanks for this library.

takkaO, 2023-01-15:
Hi @Ramon-Leeuwe 
Thank you for using this library 😊 and sorry for late reply.

I think it is a good idea.
I will be imported into the next version.

Ramon-Leeuwe, 2023-01-15:
Thanks for fixing this issue. :thumbsup: 
I really like your project, but struggled a bit with memory management (not enough memory). 
That's why I raised this issue.

closed 2023-01-15 by Ramon-Leeuwe