takkaO/OpenFontRender #18: Need some way to center text on screen

MikeyMoMo, 2023-01-13:
You don't honor or allow setTextDatum so I am left without any way to center my string on the display.  Can you tell what calls to make to enable that?  When I am using bit coded text, I choose the centered Datum and it does it for me.  I cannot figure out how to make cdrawString work.  I can't figure out how to code the Layout item to get past the compiler.  I am not even sure that will work for me since there is no specification of the size to center in.  Maybe it uses the display width.  Surely would like a valid example.  Your samples are very basic and leave out a lot.

I am sure I am missing the obvious.

martinberlin, 2023-01-13:
Hi @MikeyMoMo ,

if you check #5 you can use the getTextWidth / height to calculate this and center the text yourself making a function where you send (x, y, text) and then you can calculate it.

MikeyMoMo, 2023-01-14:
Martin,

Thank you.  I have used getTextWidth in the past but a lot of time, it has not been available and I forgot to try it this time.  Thanks for reminding me.  That is working.  At least, it is returning an answer and it is trivial to center the text from here.  

However, I would still like how to make cdrawString work.  Any example of a valid call?  I cannot figure out how to code that last parameter, the Horizontal or Vertical emum value.  It is probably bleedingly simple but I just can't make anything pass muster!

Mike

takkaO, 2023-01-15:
Hi @MikeyMoMo .
Thank you for using this library 😊 and sorry for the late reply.

For an example of centering, see [HERE](https://github.com/takkaO/OpenFontRender/blob/master/examples/M5Stack/alignment_string/alignment_string.ino) for an example of centering.

`Layout::Horizontal` or `Layout::Vertical` is used to specify the text writing direction. 
English is generally written only horizontally, but some languages, such as Japanese, use both horizontal and vertical writing styles.
However, the `Layout::Vertical` writing option is intended for future expansion and is not currently supported.

Text alignment (centered or right-aligned) is specified in the `Align` enumeration.
```c++
enum class Align {
	Left,
	Center,
	Right
};
```
Currently only horizontal align is supported, vertical align is fixed at the top (just don't have time to implement it).
If you want the text to be perfectly centered vertically, horizontally, and vertically, you must use `calculateBoundingBox` to get the rectangular area used by the text, and the user must adjust it himself.

However, it is almost never necessary to actually use the Align enumerator.
I recommend using the following functions

For left-alignment, use `drawString` or `printf`.
For centering, use `cdrawString` or `cprintf`.
For right-alignment, use `rdrawString` or `rprintf`.

For more information on the functions, see the [API documentation](https://github.com/takkaO/OpenFontRender/blob/master/doc/API.md).

Thank you.

MikeyMoMo, 2023-01-28:
Here is a much more complete answer that I finally found by just trying stuff till it fit.

```
  FT_BBox BBOX; FT_Error FTERR;
  Drawing doDraw = Drawing::Execute;
  Align myAlign = Align::Left;
  render.drawHString("Hello\nWorld", tft.width() / 2, 0, TFT_MAGENTA, TFT_BLACK, myAlign, doDraw, BBOX, FTERR);
```

>   FT_BBox BBOX;                       // This is an emum
>   //  enum class FT_BBOX {            // Not sure of the order.
>   //    xMin,                         // Can't find the actual definition
>   //    xMax,
>   //    yMin,
>   //    yMax
>   //  };
> 
>   Drawing doDraw = Drawing::Execute;  // This is an emum
>   //  enum class Drawing {
>   //    Execute,
>   //    Skip
>   //  };
> 
>   Align myAlign = Align::Left;        // This is an emum
>   //  enum class Align {
>   //    Left,
>   //    Center,
>   //    Right
>   //  };
>   
>   FT_Error FTERR;  // This seems to be an integer but I can't find any doc on it.
> 

I have not figured out how FT_Error is defined.

takkaO, 2023-05-01:
More detailed alignment settings are available in `v1.1`. 
Try using the `setAlignment()` function. 
See [GitHub Pages](https://takkao.github.io/OpenFontRender/group__layout__api.html#ga7d72198aa32b966aa60c7cdf54c83d1a) for more information.

MikeyMoMo, 2023-06-02:
cprintf works.  There are still problems with length calculations in many places.

closed 2023-06-02 by MikeyMoMo