takkaO/OpenFontRender #21: tft_eSPI printf not using bg colour

beng2k, 2023-02-01:
Hello there,

I am trying to update text in a sprite. The numbers are displayed in the right position and using the same setCursor position. The numbers are not clearing despite setting the background colour to match the background.  Here is a code snippet of the updating of the text.  I cant see what I could be doing wrong? 

```
    ofr.setDrawer(DATA3gauge);
    ofr.setFontColor(TFT_ORANGE, TFT_BLACK);
    ofr.setFontSize(35);
    ofr.setCursor(79,55);
    ofr.cprintf("ERR");
    DATA3gauge.pushSprite(320,0);
```

takkaO, 2023-02-02:
Thank you for using this library 😊

Unfortunately, I have never used the sprite feature.
Please tell me the smallest code that can reproduce the issue in my environment.
And what microcontroller you are using.

Thank you.

MikeyMoMo, 2023-02-09:
I have noticed the same thing.  Just throw something on screen with foreground and background colors set.  The background color is ignored.  If you want it, that is a very good thing == an invisible background color not changing the background of the characters.  However, if you want it to happen, it is not good the way it is.

Here is a code snippet I wrote to check out almost everything.  I posted elsewhere with other errors than came up while writing and testing this program such as text height is wrong and the bounding box is wrong.

Here, the background color setting (done 2 ways) is ignored anyway.  The letters come out white on black, not red.

```
  tft.fillScreen(TFT_BLACK);
  // Alignment Left
  // If the second font color is missing, it will be written invisible and the background color
  //  already on the screen will not be changed.  To change it, use a second color like this:
  render.setFontColor(TFT_WHITE, TFT_RED);  // Foreground, Background (optional)
  // Note, it can also be set by 8 bit hex values.  Foreground only or foreground and background, as above.
  //  render.setFontColor(0x50, 0x70, 0x150); // Mostly blue, a little green and less red. FG only set.
  //  ...or...
  //  render.setFontColor(0x50, 0x70, 0x150, 0x150, 0x70, 0x50);  // Set FG and BG to custom colors.
  //
  // You can also set the background color, only, thus:
  render.setBackgroundColor(TFT_RED);  // Change BG color only, leave font color as already set.
 
  // So far, the background color setting does nothing.  Maybe this is by design.

  // Alignment left
  render.setCursor(0, 0);
  render.printf("Hello\nWorld size ");  // Only foreground color honored.

```

beng2k, 2023-02-09:
As above. Sorry for the delay. 
I can pull some code but this above looks very similar to what I have done as well. 

As above it seems as though the background colour is ignored

MikeyMoMo, 2023-02-09:
I just uploaded my full demo code that shows this (not) happening.  It also shows how use all but the most trivial calls to this library.  It is my first Github repository.  Please be kind... ;-))  
It is here: 
> https://github.com/MikeyMoMo/OpenFontRender-Example-program

There is a notation that the font BG color is not being used.  I am not sure if that is in a sprite but it fails.  I could do it to a sprite if necessary.  I use OFR for both sprites and background (tft) screen writes and it operates the same both ways.  Also under MikeyMoMo is a very full featured program that uses OFR for almost everything, sprites and tft writes.  All works pretty well.  Just a few little bugs to work out.  SO glad to have this library!!!

takkaO, 2023-05-01:
From v1.1, you can select background fill method. 
Try using the `setBackgroundFillMethod()` function. 
See [GitHub Pages](https://takkao.github.io/OpenFontRender/group__layout__api.html#gab438b1f7d6b2afa198e9465b7d8151cd) for details.

I am really sorry for the late update🙇‍♂️🙇‍♂️🙇‍♂️

MikeyMoMo, 2023-06-02:
This works.  The lines I used are:

```
      scrollSprite.fillRect(trackTimeDate, 0, scrollSpriteW, scrollSpriteH, DarkBlue);
      ofr.setDrawer(scrollSprite);
      ofr.setFontColor(TFT_WHITE, DarkBlue);
      ofr.setBackgroundFillMethod(BgFillMethod::Block);

```