takkaO/OpenFontRender #22: FreeType Cache just keeps growing

GlovePuppet, 2023-02-06:
I have an application with a lot of strings at different font sizes. All the text is rendered using the same font. 

As I walk through my menu system I see memory being allocated as more glyphs are cached by FreeType's cache manager. Eventually all the available heap is allocated and the device resets.

I can work around this by unloading/loading the font every time I update text on the screen but it's not a great solution. Rendering speed is not noticeably worse which makes me wonder if the cache is required at all?

Is there a better way to flush free type's caches or reduce/control its memory usage??

takkaO, 2023-02-06:
Hi @GlovePuppet 
Thank you for using this library 😊

This issue has been reported as #17.

Currently there is no limit on the cache size, but it will be set to a minimum in the next update. 
The develop branch already has this fix in place.
Please wait a little longer for the master branch to be updated.

You can also use the following API to change the cache size.

```c++
void setCacheSize(unsigned int max_faces, unsigned int max_sizes, unsigned long max_bytes);
```

Be careful, this API **must be called before the `loadFont` method.**

Thank you.

GlovePuppet, 2023-02-10:
Thank you, this has solved my problem. 

Also, thank you creating this project, it really improves the "look" of my text display

takkaO, 2023-05-01:
The master branch has been updated to `v1.1`. 
The default cache size has been set to the minimum.

I am really sorry for the late update🙇‍♂️🙇‍♂️🙇‍♂️