takkaO/OpenFontRender #24: Feature request: output to image file

pkzr15, 2023-03-10:
Hello. Can the function of outputting to a image file be implemented, so that even without using actual hardware, the ttf rendering can be checked? 

Thanks.

takkaO, 2023-03-12:
Hi @pkzr15 .
Thank you for using this library 😊.

Image output functionality would not be implemented in this library.
You can use M5LcdEmulator to check the rendering results, but this project is at the alpha version level and requires some skill to use.
I plan to evolve M5LcdEmulator to make OpenFontRender more useful, but it will be a while. 
Sorry.

P.S.
I am using M5LcdEmulator to develop this library.

[image]

pkzr15, 2023-03-13:
Thanks @takkaO .

I purchased a M5Stack after asking the above question 😀 and managed to get OpenFontRender up and running.

I'm interested in implementing complex script (Hindi for example) rendering on embedded platforms,  which at the moment is not being rendered by OpenFontRender correctly, very unfortunate. The glyphs were beautiful but the order of the glyphs were not taken care of.

Anyway thanks for your kind reply and have a great day.

takkaO, 2023-05-01:
I am really sorry for the late reply🙇‍♂️🙇‍♂️🙇‍♂️

> The glyphs were beautiful but the order of the glyphs were not taken care of.

Very interesting. More information is needed to resolve this issue.
Can you give us at least the following three pieces of information?

- Expected output
- Current output
- Target font file

Lastly, unfortunately I do not understand Hindi at all 😱