takkaO/OpenFontRender #30: Feature request: loading fonts from SPIFFS/LittleFS

marcelstoer, 2023-05-08:
Maybe I didn't quite get it yet but from what I understand OpenFontRender can only load fonts from SD card or binary (i.e. embedded in code). 

Working with ESP32 it would be most helpful to be able to load fonts from the internal LittleFS file system.

infrafast, 2024-01-22:
Hello I have same understanding and need. Loading from LittleFS Thanks!

infrafast, 2024-02-12:
Here is the code I've used:

      // Open the font file
      File fontFile = LittleFS.open("/myfont.ttf", "r");
      if (!fontFile) {
        DEBUG_PRINT("Failed to open icon file");
      }

      // Read font file contents into a buffer
      size_t fileSize = fontFile.size();
      uint8_t* fontBuffer = (uint8_t*)malloc(fileSize);
      fontFile.read(fontBuffer, fileSize);
      fontFile.close();
      if (iconOfr.loadFont(fontBuffer, fileSize)) { 
        DEBUG_PRINT("\nIcon Initialise error"); 
      }

hope this helps

takkaO, 2024-05-08:
@marcelstoer , @infrafast 

OpenFontRender v1.2 now includes enhanced support for file systems.  
Font loading from SPIFFS is also supported and can be enabled simply by using a preset.   
Unfortunately, no LittleFS presets are provided yet, but it should be possible to support the LittleFS file system by implementing custom code.
Please see [document](https://takkao.github.io/OpenFontRender/md_manuals_pages.html)  and [sample code](https://github.com/takkaO/OpenFontRender/tree/master/examples) for more info.
Welcome your feedback.

Sorry for the crazy slow response.

infrafast, 2024-05-27:
Hi thanks a lot, I suggest you enapsulate the code I provided for littleFS in a method of your class then

giri-v, 2025-01-12:
I tried this and while:
SPIFFS.exists("/Roboto.ttf") returns true
ofr.loadFont("/Roboto.ttf") fails with FT_Stream_Open: could not open `/Roboto.ttf'

I am using TFT_eSPI on an ESP32 (although I'm not sure that is relevant)

takkaO, 2025-01-12:
@giri-v 

Could you share me your code?

giri-v, 2025-01-15:
Here it is...
Init Code
`void initFS()
{
#ifndef LittleFS
    // Initialize SPIFFS
    if (!SPIFFS.begin(true))
    {
        Log.errorln("An Error has occurred while mounting SPIFFS");
        return;
    }
#else
    if (!LittleFS.begin())
    {
        Log.errorln("Flash FS initialisation failed!");
        while (1)
            yield(); // Stay here twiddling thumbs waiting
    }
    Log.infoln("Flash FS available!");

#endif
}
`
here is the code that executes in setup()
`    ofr.setDrawer(tft);
    // ofr.loadFont(Roboto, sizeof(Roboto));
    if (SPIFFS.exists("/Roboto.ttf"))
    {
        Log.infoln("Loading font from file.");
        if (ofr.loadFont("/Roboto.ttf"))
        {
            Log.errorln("Failed to load font.");
            ofr.loadFont(Roboto, sizeof(Roboto));
        }
    }
    else
    {
        Log.errorln("Failed to load font.");
        ofr.loadFont(Roboto, sizeof(Roboto));
    }
`

I'm re-checking the validity of the data and partition just to be sure.

Thanks.

takkaO, 2025-01-16:
@giri-v 

Thank you share your code😄
Perhaps this is because the code for file access has not been constructed correctly.

Since access to the file system varies greatly from microcontroller to microcontroller, it is necessary to implement low-level functions for each microcontroller.
OpenFontRender provides several implementations as presets.

See [examples/M5Stack/load_from_spiffs](https://github.com/takkaO/OpenFontRender/blob/master/examples/M5Stack/load_from_spiffs/load_from_spiffs.ino) if you want to use SPIFFS with ESP32.

Unfortunately, there are no presets for LittleFS yet, so you have to create your own low-level functions. Please refer to [examples/M5Stack/load_from_sd_without_preset](https://github.com/takkaO/OpenFontRender/blob/master/examples/M5Stack/load_from_sd_without_preset/load_from_sd_without_preset.ino) for implementation.
( If you can provide us with a preset version of the LittleFS code, we will gladly merge it in. )

For a more detailed explanation, please refer to the [documentation](https://takkao.github.io/OpenFontRender/md_manuals_pages.html).

Also note that even if the implementation is correct, if the font file size is large, it may not be possible to load the file due to insufficient memory in the microcontroller.

giri-v, 2025-01-16:
LittleFS code was provided just for completeness. It (LittleFS) is NOT defined in this case, I can confirm I am using SPIFFS as I can access other files normally. I'll try again.

takkaO, 2025-01-16:
@giri-v 

I understood that you are not using LittleFS.
Any further discussion would require you to share me minimal code that can reproduce the problem.