takkaO/OpenFontRender #32: How can an integer be printed without sprintf

MikeyMoMo, 2023-06-02:
I have tried 6 different ways.  I am trying to avoid sprintf, if possible and have some kind of conversion in the single line of code.

Here's my last attempt of the 6 ways I tried:

ofr.printf(char*(timeinfo->tm_mday));  // timeinfo->tm_mday points to an integer.

gives this error:

C:\Arduino\@ESP32 Sketches\TTGO T4 v1.3\Triple_Time_on_T4_v1.3_Ver_12_MultiAC\Analog_Clock.ino: In function 'void showCorners(time_t)':
Analog_Clock:183:14: error: expected primary-expression before 'char'
   ofr.printf(char*(timeinfo->tm_mday));

also tried:

ofr.printf(char(timeinfo->tm_mday));

and several others.

what does work:

strftime(charWork, sizeof(charWork), "%d", localtime(&now));  // Numeric day of month
ofr.setCursor(0, tft.height() - 26); ofr.printf(charWork);
  
But that takes strftime or sprintf.  Was trying for some kind of inline conversion like can be used practically everywhere else in ESP land.

Mike

takkaO, 2024-05-08:
@MikeyMoMo 

Is the following not good enough?

```c
ofr.printf("%d", timeinfo->tm_mday);
```

MikeyMoMo, 2024-05-08:
This was asked about a year ago.  I mostly dropped this due to extreme slowness.  I could not afford the time in time-critical functions.  If I ever get back to try it for something, again, I will see what can be done.  This was not an answer to the question, actually, but never mind, I don't really care much any more.

closed 2024-05-08 by MikeyMoMo