takkaO/OpenFontRender #33: Nucleo64 : undefined reference to `_open'

CFTechno, 2023-06-08:
c:/users/xxx/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: 

c:/users/xxx/appdata/local/arduino15/packages/stmicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(lib_a-openr.o): in function `_open_r':
(.text._open_r+0x10): undefined reference to `_open'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

wystewart, 2023-07-07:
you need to modify filesupport.cpp and .h for your board.  The library only supports M5STack and WIOTerminal at the moment.  From wiki:

SD card access is strongly hardware dependent, so for hardware other than M5Stack and Wio Terminal, you will need to add file manipulation functions to [FileSupport.cpp](https://takkao.github.io/OpenFontRender/_file_support_8cpp.html)/.h.

Uzair835, 2024-01-16:
I am also having this issue and unsure how to go about fixing it. I am using an STM32 (bluepill), and NOT loading the font from an SD card. How can the library be modified to accept the font file being saved within the chip? Everything that I see appears to rely on FileSupport and loading files from the SD card

takkaO, 2024-01-16:
I'm implementing changes to handle the file I/O processes that different for each board in an unified manner. There are in the tmp branch now.

Users can handle all types of storage by simply providing five functions (`fopen`, `fclose`, `fread`, `fseek`, `ftell`) to manipulate files from any storage.

The input/output and operation of these five functions are the same as standard C language functions.

I plan to integrate it into the master branch in the near future.
Thank you.

Uzair835, 2024-01-16:
Thank you! At the moment, is there any workaround to using this library without an SD card? I see examples about loading binary fonts, but even when copying those I get the same error of there being no reference to _open. Just want to make sure I am not missing something obvious.

takkaO, 2024-01-17:
@Uzair835 

Perhaps the `fopen` function etc. is not recognized, try including stdio.h in `FileSupport.h`.
If including `stdio.h` causes other problems, try defining an appropriate empty function in `FileSupport.cpp`.

takkaO, 2024-05-08:
@Uzair835 

OpenFontRender v1.2 now includes enhanced support for file systems.  
This should allow font files to be loaded on various hardware without library changes.  

Please see [document](https://takkao.github.io/OpenFontRender/md_manuals_pages.html)  and [sample code](https://github.com/takkaO/OpenFontRender/tree/master/examples) for more info.
Welcome your feedback.

Sorry for the crazy slow response.