takkaO/OpenFontRender #34: STM32H750

wystewart, 2023-07-06:
not really an issue, but I have a STM32H750 board running at 480MHz, SDRAM at 120MHz with STM32duino (code needed some modification to compile for file system).  I thought I'd pass along the render speed of your demo code.  Using only drawpixel, 65ms.  This is with a modified version of Adafruit_GFX, to a 640x480 LCD running in 32 bit pixels (ARGB).  The fonts look fantastic. thanks so much for developing this library.

takkaO, 2024-05-08:
Thank you for using this library 😄

I will add the information you gave us to the README in `Tested hardware`.

Also, OpenFontRender v1.2 now includes enhanced support for file systems.  
This should allow font files to be loaded on various hardware without library changes.
Please see [document](https://takkao.github.io/OpenFontRender/md_manuals_pages.html)  and [sample code](https://github.com/takkaO/OpenFontRender/tree/master/examples) for more info.
Welcome your feedback.

Sorry for the crazy slow response.

wystewart, 2024-05-09:
thank you.  And you did acknowledge my post with a thumbs up, so no apology needed for "crazy slow response".  Go ahead and close this ticket since it was informational only.

closed 2024-05-09 by takkaO