takkaO/OpenFontRender #35: Using SPIFFS instead of SD

PrabhieG, 2023-08-28:
How can I use SPIFFS to directly access files flashed onto the ESP32 instead of SD? The library works when using a header created from ttf2bin.py. But doesn't work using uploaded file to SPIFFS, due to no SD card. I tried going in FileSupport.cpp and changing `fs::FS &fontFS = SPIFFS;` from `fs::FS &fontFS = SD;`. But still there is an error:
`FT_Stream_Open: opened `/font.ttf' but zero-sized`
font.ttf is not zero-sized and I can confirm by using file.read() which works.

takkaO, 2024-05-08:
OpenFontRender v1.2 now includes enhanced support for file systems.  
Font loading from SPIFFS is also supported and can be enabled simply by using a preset.   
Please see [document](https://takkao.github.io/OpenFontRender/md_manuals_pages.html)  and [sample code](https://github.com/takkaO/OpenFontRender/tree/master/examples) for more info.
Welcome your feedback.

Sorry for the crazy slow response.

PrabhieG, 2024-05-08:
All good thanks for the reply! Sadly I did move onto other things and no longer have the ESP to test with, and I can't recall any longer what I ended up doing but it was working. Awesome library, thank you.

closed 2024-05-08 by PrabhieG