takkaO/OpenFontRender #42: Vertical centering with a line break

larrybud2004, 2023-12-17:
I have a function which draws simple large text on the screen, e.g.

`
        ofr.setCursor(tft.width() / 2, tft.height() / 2);

	ofr.setFontColor(GLT_COLOR_GOLD, TFT_BLACK);

	ofr.setFontSize(46);

	ofr.setAlignment(Align::BottomCenter);

	ofr.printf(label);
`

This works great for a single line of text:
[image]

But if my "label" has a line break, e.g. \n, the vertical alignment is not correct:

[image]

Am I doing this wrong or does the library not support this kind of thing?

scottchiefbaker, 2023-12-18:
You may be running in to the same thing I'm seeing in #38 ?

larrybud2004, 2023-12-18:
Note sure it's the same issue directly, but perhaps related in how he calculates the text height? With my example, the single line is centered properly, however I have to set it to bottom alignment for it to look right. Might have just gotten lucky with this particular font size that bottom alignment is centered. If I choose Middle alignment, it ends up too low, which is what you're experiencing.

The alignment of top/middle/bottom is not intuitive at all, and I'm not clear on what it's supposed to be aligning _to._

I just have a feeling the height calculation is being done incorrectly.

scottchiefbaker, 2023-12-18:
You can see a better visual example of the text alignment stuff with the code in [this PR](https://github.com/takkaO/OpenFontRender/pull/40).

I suspect you are correct, and there is a minor bug in glyph height calculation. I tried looking at the [alignment code](https://github.com/takkaO/OpenFontRender/blob/master/src/OpenFontRender.cpp#L597) but it went over my head pretty quick.

larrybud2004, 2023-12-18:
Yeah, unfortunately I don't know much about how TTF are stored, or much about typesetting.

One thing that threw me for a minute when trying to set text centered in a box is when lower case letters were used that had a descender (g, y, q).

Not sure if that's related, but when trying to use the Middle alignment it didn't work as I expected it, as the descender was obviously included in the calculation for the height of the character.

For example, try aligning in the middle "Start" and "Play" and you'll see two different results because of the lowercase "y".

mavriq-dev, 2025-01-21:
I have created a fix [here](https://github.com/takkaO/OpenFontRender/pull/49).

takkaO, 2025-06-02:
#49

closed 2025-06-02 by takkaO