takkaO/OpenFontRender #43: Recommendations for memory use?

chconnor, 2024-01-14:
Hello!

I noticed that my SRAM drops by about 4k after a single myofr.printf() call (drawing a time to the screen). However, if I repeat that call many times, the SRAM only drops once.

So I assume that it is just a cache in OpenOFR that is caching something the first time?

Are there any recommendations you can give to protect SRAM? E.g.:

- is it smart to use the same sized font to avoid needing to render it at different sizes?
- does it cache per glyph, so that using more varied glyphs will use more memory?
- do larger font sizes mean more memory use?
- can the cache be cleared periodically to save SRAM?
- alternately, can the cache be forced to grow to maximum size to guarantee a bound on memory usage?

Thanks!

takkaO, 2024-05-08:
Hello @chconnor 

Depends on the behavior of the FreeType2 library.
You can change the size to cache with `setCacheSize`, but it simply maps the FreeType2 library.

Please see [FreeType2 FTC_Manager_New](https://freetype.org/freetype2/docs/reference/ft2-cache_subsystem.html#ftc_manager_new).

I am not aware of all the specific behaviors, but perhaps the answer is as follows.

> is it smart to use the same sized font to avoid needing to render it at different sizes?

Probably not necessary

> does it cache per glyph, so that using more varied glyphs will use more memory?

Probably YES.

> do larger font sizes mean more memory use?

YES, but temporary increase in memory usage for drawing, not caching

> can the cache be cleared periodically to save SRAM? (Edit: I see setCacheSize() exists to limit cache size)

Conditionally possible.
FreeType provides FTC_Manager_Reset and OpenFontRender calls it in unloadFont. Current implementation requires font loading again.

> alternately, can the cache be forced to grow to maximum size to guarantee a bound on memory usage? (Edit: I see setCacheSize() exists to limit cache size)

Use setCacheSize. By default it is set to the minimum cache size.

Thank you.
Sorry for the crazy slow response 🙇‍♂️🙇‍♂️🙇‍♂️

chconnor, 2024-05-08:
Appreciate the reply!

closed 2024-05-08 by chconnor