takkaO/OpenFontRender #44: Memory issue

copantok, 2024-02-11:
Hello, I am using a OpenFontRender on ESP32 together with TFT_eSPI and I am using multiple fonts. I noticed that after loadFont() and then unloadFont(), approximately 50 bytes of memory remain allocated. So after each loadFont() - unloadFont() of another font, the memory usage increases until the system crashes.

llapp0612, 2024-02-28:
Can confirm memory leaks with loadFont() in a loop till system crashing an reseting.

takkaO, 2024-05-08:
@copantok , @llapp0612 

Where do you load fonts from?
Also, can you provide the smallest complete code that can reproduce the problem?

Thank you.

llapp0612, 2024-05-08:
it have nothing to do with the font itself.

when you use doubles time loadFont() and use unloadFont(), you will get a memory leak. i had posted the smallest code. 

this is incorrect:

void function1(){
loadFont();
<-- any code -->
function2();
<-- any code -->
unloadFont();
}

void function2(){
loadfont();
drawFont();
unloadFont();
}

in this example, load font will be load, then a call to function2 with loadFont again without unloadFont before. after second time loadFont comes unloadFont and a second time again. this is not working and will get a memory leak. after loadfont, you will need to unloadFont again, after that u can call the LoadFont again.

takkaO, 2024-05-09:
@llapp0612 

Please tell me a complete sketch (code) that can be executed by copy-paste to accurately verify the issue.
I have some overall picture with your minimal code, but I do not understand some of the details.

- What microcontroller is used?
- Is the development environment Arduino?
- Are the fonts to be loaded the same? Are they different?
- Are the fonts loaded from SD?
- Is there one instance of OpenFontRender? More than one?
- Where is function1 called from? setup or loop?

Thank you.

llapp0612, 2024-05-10:
Here is my [Git](https://github.com/llapp0612/ESP-Hardware-Monitor)

**What microcontroller is used? Is the development environment Arduino?**
First, have coded with a Arduino board, later switched to ESP32 Dev Board and changed a bit code for ESP32.

**Are the fonts to be loaded the same? Are they different?**
If you look into my code, you will see there is no different.

**Are the fonts loaded from SD?**
No. Not tested from SD.

**Is there one instance of OpenFontRender? More than one?**
I use many Times loadFont() and after that unLoadFont(), but i never use loadFont() loadFont() and then unloadFont() unloadFont(), because, this is the reason for memory leak.

**Where is function1 called from? setup or loop?**
Yes, in loop