takkaO/OpenFontRender #46: Recommendations for code changes

wystewart, 2024-05-09:
Some global variables in the code many affect having multiple instances of the fontrender.
in my copy I moved
   FT_Library g_FtLibrary;
   bool g_NeedInitialize
from OpenFontRenderer.cpp into the the private section of the class OpenFontRender in the .h file.

I did not check if any of the underlying code has globals.

takkaO, 2024-05-09:
@wystewart 

> Some global variables in the code many affect having multiple instances of the fontrender.

Can you explain in detail how this will affect?

If I understand correctly, FT_Library does not need to be independent in each instance.
If the problem is that global variables are ugly, redefining them as class variables (static members) might be a good idea.

wystewart, 2024-05-10:
Ok, I assumed that FT_Lbrary needed to be independent for each instance and since g_NeedInitialize is protecting FT_Lbrary the same could be said for that too.  I'll do some tests when I have a chance.