takkaO/OpenFontRender #51: check if a glyph exists

infrafast, 2025-02-14:
Hello, I experience some crash most likely because I try to render char that is not in the font file.
I was wondering if there is a way to check if a glyph exists in the font or other mechanism to prevent displaying someting which do not exists ?
Thank you !
BR

takkaO, 2025-02-15:
@infrafast 

In my environment, when I try to output a non-existent character, a dot character (・) is displayed. Please share me the font file you are using and a minimal program that can reproduce the problem.

Thank you.

infrafast, 2025-02-27:
It's very difficult to reproduce, actually I output all characters 
using following string:

"!\"#$%&'()*+,-./ 0123456789 :;<=>?@ ABCCDEFGHIJKLMNNOPQRSTUVWXYZ [\\]^_` abcdefghijklmnopqrstuvwxyz {|}~çèéà"

it doesnt crash but it doesnt render the % sign which I guess maybe the reason of my issue. Here is the font file enclosed and a picture, would you have any idea why ?

[image]

takkaO, 2025-03-05:
@infrafast 

I will check in my environment, please give me some time.

takkaO, 2025-03-20:
@infrafast 

I don't know your code so I could be wrong, but aren't you using the printf method?
The printf method supports C format specifiers, so the % must be escaped.

```c++
render.printf("%%"); 
```

and no error occurs in my environment.

infrafast, 2025-03-27:
thanks actually I am using cprintf  to display a string of character, not characters individually

render->cprintf(value);

so it means I will have to parse the value and replace all % by %%, right ?

takkaO, 2025-03-28:
@infrafast 

Probably.
Any further support requires a minimum program code that can reproduce the problem.

infrafast, 2025-03-28:
having %% solved the issue!
Thanks for your support ! We can close this issue